ArduPilot/ardupilot_gz #1: Set up basic infrastructure for running a CLI smoke test

Ryanf55, 2023-03-22:
Create a ROS2 package that can run a smoke test to ensure the DDS functionality in ardupilot is working as intended.
 

- [ ]  Can be build with `colcon build`
- [ ] Can be tested with `colcon test`
- [ ] A failure to receive any messages results in failure
- [ ] A failure to receive messages within the expected rate results in failure

srmainwaring, 2023-04-06:
Basic CLI smoke tests will probably go into the main ArduPilot repo. Development is on this branch: https://github.com/srmainwaring/ardupilot/tree/wip/ros2-pkg-cmake-v2.

## Status

- A PoC testing the time message is published is running on the docker image https://github.com/ArduPilot/ardupilot_dev_docker/blob/ros/docker/Dockerfile_dev-ros

Ryanf55, 2023-05-05:
Closed by #23429

closed 2023-05-05 by Ryanf55