ArduPilot/ardupilot_gz #2: Support launching gazebo sim to visualize vehicle

Ryanf55, 2023-03-22:
- [ ] Works with ROS2 Humble
- [ ] Works with Gazebo Garden
- [ ] If you launch sitl with copter, a copter shows up in gazebo
- [ ] If you launch sitl with a plane, a plane shows up in gazebo
- [ ] If you move the vehicle in sitl, it moves in gazebo
- [ ] Each vehicle has an attached camera output with a corresponding ros2 camera topic

srmainwaring, 2023-04-15:
There is an initial version of a bring-up package for an ArduPilot / microROS DDS / ROS 2 / Gazebo system here:

- https://github.com/srmainwaring/ardupilot_ros2/tree/wip/ardupilot-gz

The system can be brought up by a single launch file:

```bash
ros2 launch ardupilot_gz_bringup bringup_iris.launch.py
```

Run a interactive MAVProxy session:

```bash
mavproxy.py --console --map
```

The topics published by ArduPilot may be inspected with:

```bash
ros2 topic list
/ROS2_BatteryState0
/ROS2_NavSatFix0
/ROS2_Time
/clicked_point
/clock
/goal_pose
/initialpose
/iris/odometry
/joint_states
/parameter_events
/robot_description
/rosout
/tf
/tf_static
```

srmainwaring, 2023-04-16:
Moved code to new repo: https://github.com/srmainwaring/ardupilot_gz/pull/1

Ryanf55, 2023-09-04:
> Moved code to new repo: https://github.com/srmainwaring/ardupilot_gz/pull/1
> 
> ### Docker image
> Notes for running the `ardupilot_gz packages` on the docker image [`Dockerfile_dev-ros`](https://github.com/ArduPilot/ardupilot_dev_docker/blob/master/docker/Dockerfile_dev-ros).
> 
> The image is configured for CI and requires additional packages to be used with Gazebo (in server only mode).
> 
> #### Summary
> * [ ]  Install Gazebo Garden.
> * [ ]  Install `libsdformat12-dev`.
> * [ ]  Install `rapidjson-dev`.
> * [ ]  Install `libgflags-dev`, `ros-humble-ament-cmake-mypy`, `ros-humble-image-transport`, `ros-humble-image-transport-plugins`.
> 
> #### Details
> Install `gz-garden` and `libsdformat12-dev`
> 
> ```shell
> sudo wget https://packages.osrfoundation.org/gazebo.gpg -O /usr/share/keyrings/pkgs-osrf-archive-keyring.gpg
> 
> echo "deb [arch=$(dpkg --print-architecture) signed-by=/usr/share/keyrings/pkgs-osrf-archive-keyring.gpg] http://packages.osrfoundation.org/gazebo/ubuntu-stable $(lsb_release -cs) main" | sudo tee /etc/apt/sources.list.d/gazebo-stable.list > /dev/null
> 
> sudo apt-get update
> sudo apt-get install gz-garden
> sudo apt install libsdformat12-dev
> ```
> 
> Install dependencies for `ardupilot_gazebo`
> 
> ```shell
> sudo apt install rapidjson-dev
> ```
> 
> Install dependencies for `ros_gz`
> 
> ```shell
> sudo apt install libgflags-dev
> sudo apt install ros-humble-ament-cmake-mypy
> sudo apt install ros-humble-image-transport
> sudo apt install ros-humble-image-transport-plugins
> ```
> 
> Since ROS 2 Humble and Gazebo Garden is not an officially supported combination, `rosdep` is not able to resolve all the required dependencies:
> 
> ```shell
> rosdep check --from-paths src
> System dependencies have not been satisfied:
> apt	ros-humble-ros-gz-bridge
> apt	ros-humble-ros-gz-sim
> apt	ros-humble-ros-gz-sim-demos
> apt	ros-humble-ros-gz-image
> apt	libgflags-dev
> apt	ros-humble-ament-cmake-mypy
> apt	ros-humble-ament-cmake-pclint
> apt	ros-humble-ament-cmake-pycodestyle
> apt	ros-humble-image-transport
> apt	ros-humble-image-transport-plugins
> apt	ros-humble-rqt-image-view
> apt	ros-humble-rqt-plot
> apt	ros-humble-rqt-topic
> apt	ros-humble-rviz2
> apt	ros-humble-sdformat-urdf
> apt	ros-humble-xacro
> apt	ros-humble-sdformat-test-files
> apt	ros-humble-gps-msgs
> apt	ros-humble-ros-gz-interfaces
> ERROR[ros_gz_sim]: Cannot locate rosdep definition for [gz-math7]
> 	rosdep key : gz-math7
> 	OS name    : ubuntu
> 	OS version : jammy
> 	Data:
> <no data>
> ERROR[ros_gz_image]: Cannot locate rosdep definition for [gz-transport12]
> 	rosdep key : gz-transport12
> 	OS name    : ubuntu
> 	OS version : jammy
> 	Data:
> <no data>
> ERROR[ros_gz_sim_demos]: Cannot locate rosdep definition for [gz-sim7]
> 	rosdep key : gz-sim7
> 	OS name    : ubuntu
> 	OS version : jammy
> 	Data:
> <no data>
> ERROR[ros_gz_bridge]: Cannot locate rosdep definition for [gz-transport12]
> 	rosdep key : gz-transport12
> 	OS name    : ubuntu
> 	OS version : jammy
> 	Data:
> <no data>
> ```

The fact that the GZ rosdep keys will never resolve is actually pretty annoying. If we want to run this in CI, we could add the --skip-keys arguments, and then remove `-r` to enfoce our package.xml's work here.

srmainwaring, 2023-09-20:
Closing: we have not added a plane, but the two copter examples demonstrate the procedure for adding further vehicles.

closed 2023-09-20 by srmainwaring