ArduPilot/ardupilot_gz #4: Add a debian install of the ROS2 tools

Ryanf55, 2023-04-22:
On Ubuntu, add a launchpad.net  account for ArduPilot. 
Add versioned debs for Ardupilot and its tools. 

This will allow users to easily install debians of ArduPilot. Can click to add that project to your deb sources, and then you can update+install. 

* ardupilot_sitl
* ardupilot_gz
* ardupilot_dds

The prebuilt binaries are avilable for stable, 4.4, latest, etc. 

Can look into having OSRF have the ardupilot repo on the build farm. We would want to make available ARdupilot4.4, ArdupilotRolling, etc.

pulak-gautam, 2024-02-09:
Hi, can you assign me this issue? I believe I have requisite skills, and have experience doing a similar thing. 
Thank you

Ryanf55, 2024-02-09:
That would be great! Let me know if you have questions.

pulak-gautam, 2024-02-27:
Hi, ``ardupilot_gz_bringup`` seems to depend on ``ardupilot_sitl`` and ``ardupilot_msgs``. These are located under ardupilot/Tools/ros2. Do they have their own github like ``ardupilot_gazebo`` or other packages?
I would be needing to add the dependencies on my custom rosdistro for rosdep to find it - while running bloom-generate. I will try to find a workaround.
Thanks

Ryanf55, 2024-02-27:
> Hi, `ardupilot_gz_bringup` seems to depend on `ardupilot_sitl` and `ardupilot_msgs`. These are located under ardupilot/Tools/ros2. Do they have their own git remote like `ardupilot_gazebo`? Thanks

Nope, they are local builds.

pulak-gautam, 2024-03-05:
Hi, so I did manage to get the debs built however since ``ardupilot_sitl`` has dependency on ``micro_ros_agent``, this breaks installation on any system without the custom rosdistro index (i.e. without micro_ros_agent added in there) see [1](https://github.com/micro-ROS/micro-ROS-Agent/issues/176).
Is ``micro_ros_agent`` a hard dependency of ``ardupilot_sitl``? Is there a way for me to circumvent around it?
Thanks

Ryanf55, 2024-03-05:
> Hi, so I did manage to get the debs built however since `ardupilot_sitl` has dependency on `micro_ros_agent`, this breaks installation on any system without the custom rosdistro index (i.e. without [...]

It is a hard dependency. I would ask around on the microROS slack channel to see if it's in the roadmap to get debians built.

Ryanf55, 2024-03-05:
Yea, check it out: https://docs.vulcanexus.org/en/latest/rst/introduction/metapackages.html#vulcanexus-micro

pulak-gautam, 2024-03-06:
Hi @Ryanf55 , it works and I am done with generating debs for all the packages that you have listed on the ticket (will make a PR after testing it out on a docker container).
I had just one doubt regarding ``ros_gz`` packages. Currently, I believe we are building ``ros_gz_bridge`` and ``ros_gz_sim`` from source. However, I believe there are binary install available for ``humble`` and its ``gz-garden`` bindings (see [here](https://gazebosim.org/docs/garden/ros_installation#using-an-specific-gazebo-version-with-ros-2))
> Gazebo Garden can be used with ROS 2 Humble and non ROS official binary packages hosted in packages.osrfoundation.org. These packages conflict with ros-humble-ros-gz* packages (Humble officially supports Gazebo Fortress).
  To install the binary Gazebo Garden/ROS 2 Humble packages:
 Folow [these instruction to install gz-garden](https://gazebosim.org/docs/garden/install_ubuntu#binary-installation-on-ubuntu) from [packages.osrfoundation.org](https://packages.osrfoundation.org/gazebo/ubuntu/) repository.
      Install ros_gz from the non official binary packages from apt:
          ``apt-get install ros-humble-ros-gzgarden``

Is there a reason why we are adamant about the source install of ``ros-gz``? (if that is the case, then we would have to host our binaries of ``ros-gz``, or I need to find some other workaround).

Ryanf55, 2024-03-06:
> Hi @Ryanf55 , it works and I am done with generating debs for all the packages that you have listed on the ticket (will make a PR after testing it out on a docker container). I had just one doubt regarding `ros_gz` packages. Currently, I believe we are building `ros_gz_bridge` and `ros_gz_sim` from source. However, I believe there are binary install available for `humble` and its `gz-garden` bindings (see [here](https://gazebosim.org/docs/garden/ros_installation#using-an-specific-gazebo-version-with-ros-2))
> 
> > Gazebo Garden can be used with ROS 2 Humble and non ROS official binary packages hosted in packages.osrfoundation.org. These packages conflict with ros-humble-ros-gz* packages (Humble officially supports Gazebo Fortress).
> > To install the binary Gazebo Garden/ROS 2 Humble packages:
> > Folow [these instruction to install gz-garden](https://gazebosim.org/docs/garden/install_ubuntu#binary-installation-on-ubuntu) from [packages.osrfoundation.org](https://packages.osrfoundation.org/gazebo/ubuntu/) repository.
> > Install ros_gz from the non official binary packages from apt:
> > `apt-get install ros-humble-ros-gzgarden`
> 
> I have used these^ for now. They seemed to work atleast on my local. Is there a reason why we are adamant about the source install of `ros-gz`? (if that is the case, then we would have to host our binaries of `ros-gz`, or I need to find some other workaround).

If you can get a way to work with either garden or harmonic without compiling those packages from source, please contribute a PR. That sounds nice.

pulak-gautam, 2024-03-06:
Hi, I tested it on a docker container, it seems to work.
How do you recommend should I go about making a PR for this? I have added the release .deb files on my fork. Should I be uploading those to a folder and send a PR?

pulak-gautam, 2024-03-06:
Hi @Ryanf55, I have tested all the debian installs on a docker container, they seem to work.
However, I am not sure how should I be going about making a PR with the .deb binaries? I have added them as a release on my fork [here](https://github.com/pulak-gautam/ardupilot_gz/releases/tag/v0.0.0).
Could you let me know how should I be going about it?

pulak-gautam, 2024-03-13:
Hi, I have made a shell script [here](https://gist.github.com/pulak-gautam/8f9fcf563d13eb9ca0d287766618840d), for the generation debian installs to be added to the CI. Please let me know how should I proceeding further? Should I make github-action pipeline for this and put up a PR?
Thanks

Ryanf55, 2024-03-14:
Awesome, that's perfect. I'll run this on my system, and once that's good to go, I think the next step is to follow the bloom tutorial (creating a release organization, and then running bloom to generate the binaries.)

One thing that's not clear to me is how we can tie these binaries to 4.5 ardupilot vs 4.6. If you have any ideas on versioning, let me know. 

Excellent work so far. The script is very helpful.

pulak-gautam, 2024-03-14:
The versions in generated binaries are linked to the version tag in the package.xml. Perhaps, we can suitably change those accordingly.

If you run into any trouble while running the script, please let me know, I will try to make the necessary changes. (I did make sure to test it in a docker container).
Thanks

Ryanf55, 2024-03-14:
> The versions in generated binaries are linked to the version tag in the package.xml. Perhaps, we can suitably change those. For tieing it to Ardupilot 4.5 vs 4.6, or in similar cases. I believe instead of cloning the rolling release I could download the corresponding release source code, and use that.
> 
> If you run into any trouble while running the script, please do let me know, I will try to make the necessary changes. (I did make sure to test it in a docker container). Thanks

Yep, makes sense. The thing is that ArduPilot has its own release cadence that is not tied at all to the release of ROS versions, and I haven't yet found a good recommendation on how to do that. The only other well-known tool that releases multiple versions in the ROS ecosystem at a separate cadence is gazebo (you can run ignition, garden and harmonic all in Ubuntu 22). Perhaps there's a process we could use to do that sort of thing. 

The other (simpler) option is to not worry about ABI stability for now, and just target ardupilot master.

pulak-gautam, 2024-09-27:
@Ryanf55 Could you give me an update and direction on how to get these deb binaries on-boarded, I would be more than happy to make follow-up changes to get this thing done. I believe it would be really beneficial for a quite a lot of people (including me). 
Thanks