ArduPilot/ardupilot_gz #19: Handle microxrceddsgen dependency

pedro-fuoco, 2023-06-18:
When following the README in a fresh computer microxrceddsgen was not installed like the rest of the dependencies, would it be possible to install it with vcs with the other packages? 
[image]

pedro-fuoco, 2023-06-18:
To go around it I followed the README at `ardupilot/library/AP_DDS/`

srmainwaring, 2023-06-19:
> I just want to know if there's a reason why it isn't there beforehand

The `microxrceddsgen` package was included in the initial draft of the `rosdep` file for macOS. It was dropped because it is also included in the `AP_DDS` instructions and is a java project that does not have a `CMakeLists.txt` to build with `colcon` (so you'd have to build manually in any case).

pedro-fuoco, 2023-06-20:
Fair point. I think it should be mentioned that users need to go ahead and follow this part in the `AP_DDS` instructions then. Otherwise there's no way to guess this without asking someone.
I could submit a PR to add this to the "Dependencies" part of the README, wdyt?

pedro-fuoco, 2023-07-08:
@srmainwaring what do you think?

tejalbarnwal, 2023-07-13:
> Fair point. I think it should be mentioned that users need to go ahead and follow this part in the `AP_DDS` instructions then. Otherwise there's no way to guess this without asking someone. I could submit a PR to add this to the "Prerequisites" part of the README

I second this

srmainwaring, 2023-07-24:
Closed by #22

closed 2023-07-24 by srmainwaring