ArduPilot/ardupilot_gz #35: Prefer pathlib to os.path in launch files

srmainwaring, 2023-10-25:
Update launch files to consistently use `pathlib` rather than `os.path`.

SrijaneeBiswas, 2023-12-27:
I can take this up. What would be the acceptance criteria for this issue? 

I have a Ubuntu 22.04/ROS Humble setup.

Ryanf55, 2024-01-21:
Hello - if you are still interested, a pull request is surely welcome.

Everwhere you see any call to `os.path`, replace it with `pathlib.Path` equivalent. 

As an example, see here:
https://github.com/ethz-asl/grid_map_geo/pull/56

Ryanf55, 2024-01-24:
Yes, that sounds sufficient. I'll test the changes locally before merge too.

pulak-gautam, 2024-02-06:
Hi, I have put up a pr (#47) with changes following your instructions. Kindly review the same.
Thank you for your time.

closed 2024-02-06 by Ryanf55