ArduPilot/ardupilot_gz #46: Errors on step 4

flyasky, 2024-02-04:
using ros2-humble & gz-harmonic (same with gz-garden)

on Step 4 from readme.md I've got:
```
a@a-i5  ~/ros2_ws  rosdep install --rosdistro $ROS_DISTRO --from-paths src -i -r -y

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ros_gz_bridge: Cannot locate rosdep definition for [gz-transport13]
sdformat_urdf: Cannot locate rosdep definition for [sdformat14]
ros_gz_sim: Cannot locate rosdep definition for [gz-transport13]
ros_gz_image: Cannot locate rosdep definition for [gz-transport13]
ardupilot_gazebo: Cannot locate rosdep definition for [gz-sim8]
ardupilot_gz_gazebo: Cannot locate rosdep definition for [gz-sim8]
```

srmainwaring, 2024-02-23:
Hi @flyasky, your issue sounds similar to the one raised in this discussion on Discord: https://discord.com/channels/674039678562861068/850924523610963988/1210488910896041984

The rosdep warnings can be ignored if you have installed `gz-garden` or `gz-harmonic` manually with `apt`. Some things to check:

Make sure the system is fully upgraded:

```bash
sudo apt update && sudo apt upgrade
```

Ensure that the `GZ_VERSION` is set in each terminal (preferably in `~/.bashrc`)

```bash
export GZ_VERSION=harmonic
```

Make sure you are building in `ros_ws`, not `ros_ws/src`

Run a clean build after updating system dependencies:

```bash
cd ~/ros_ws
rm -rf ./build ./install ./log
colcon build ...
```

You can also check the non-colcon build of the ardupilot_gazebo plugin works by following the instructions here: https://github.com/ArduPilot/ardupilot_gazebo

Ryanf55, 2024-04-20:
Closed due to no feedback.

closed 2024-04-20 by Ryanf55