ArduPilot/ardupilot_gz #48: Can I use it? "wind-effects-system"

GreenPine-CK, 2024-02-13:
Hello, 

Like PX4-SITL
I'd like to try applying WIND in ardupilot_gz.

Is it impossible to implement it yet?

Or

Should I copy and paste the code on the link below?

'https://github.com/gazebosim/gz-sim/blob/af73ebe7c8c693fd54e391f79c11bf9f24df2640/examples/worlds/wind.sdf'

I'm always thankful

Ryanf55, 2024-04-20:
Yea, you can add a world with wind. That would be great.

GreenPine-CK, 2024-04-20:
Hi. Can you give me some ideas on how to apply wind environment to the ardupilot gazebo?

Ryanf55, 2024-04-20:
Yea, try copy and pasting that code and seeing how it works. Set the wind super high and see if the model gets blown away.

GreenPine-CK, 2024-04-20:
Well, when I copied and pasted the code in the link above, no gusts or winds apply... I'm entering it in the .world file, right?

Ryanf55, 2024-04-20:
> Well, when I copied and pasted the code in the link above, no gusts or winds apply... I'm entering it in the .world file, right?

Gazebo harmonic doesn't use files that end in `.world`.