ArduPilot/ardupilot_gz #50: ardupilot_sitl build error with ROS 2

aarsht7, 2024-04-18:
Hello,

I am trying to test ardupilot with gazebosim (gz-harmonic) and ROS 2 (humble)
Ubunut 22.04
Following [this](https://ardupilot.org/dev/docs/ros2-gazebo.html) guide

I encountered the following error. 
Yes, I did follow the installation for the `microxrcedds_gen` 
here is the output for the version info

```bash
ros@aarsh-pc:/workspaces/ardupilot$  microxrceddsgen -version
openjdk version "11.0.22" 2024-01-16
OpenJDK Runtime Environment (build 11.0.22+7-post-Ubuntu-0ubuntu222.04.1)
OpenJDK 64-Bit Server VM (build 11.0.22+7-post-Ubuntu-0ubuntu222.04.1, mixed mode, sharing)
microxrceddsgen version: 2.0.2
```

Error while building the `ardupilot_sitl`
```bash
Starting >>> ardupilot_sitl
--- stderr: ardupilot_sitl                         
gmake[2]: *** [CMakeFiles/ardupilot_build.dir/build.make:70: CMakeFiles/ardupilot_build] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:168: CMakeFiles/ardupilot_build.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< ardupilot_sitl [1.57s, exited with code 2]
```

Is it a known issue?
I tried to look up the internet, but there was no lead. 

Thanks.

rmackay9, 2024-04-18:
This is probably for @srmainwaring or @Ryanf55 to answer.

I'm sure you've seen it but hopefully you've got the ArduPilot build environment installed on the PC as well. https://ardupilot.org/dev/docs/building-the-code.html

Ryanf55, 2024-04-19:
Are you sure there isn't more log data? Normally, in a failure, you get messages on what failed.

Ryanf55, 2024-04-19:
Please delete the build, log, install folder. Then rebuild. 
Finally, zip up the entire contents of your log folder and attach it here. I want to look at everything. 
https://colcon.readthedocs.io/en/released/user/log-files.html#log-files

srmainwaring, 2024-04-19:
@aarsht7 the top level colcon messages are not that helpful - basically telling you a package failed to build and not much else. There are flags you can set to provide more detail `--event-handlers console_direct+`.

In this case the problem is the core ardupilot library as a colcon package. You can build this directly using `./waf` as follows:

- Change to `~/ros_ws/src/ardupilot`
- Configure: `./waf configure --board sitl --debug --enable-dds`
- Build: `./waf build`

I suspect you may be missing a dependency, possible the path to the `microxrceddsgen` binary.

aarsht7, 2024-04-19:
> Are you sure there isn't more log data? Normally, in a failure, you get messages on what failed.

On the colcon build logs, there is no log data of `ardupilot_sitl` as pkg is not built
But there is `config.log` inside `src/ardupilot/build`
After analyzing it, I realized that the `$PATH` variable it is not the same as the one in the current `env`. Maybe it is catching the old one and not updating when I rebuild without removing the ardupilot build directory.

So I tried  `rm -rf src/ardupilot/build build/ install/ log/` and rebuild
This time it does update to the correct path to the `microxrcedds_gen`, But the error remains the same.

Also, I forgot to mention one more error, so I have updated the issue.

> There are flags you can set to provide more detail `--event-handlers console_direct+`.

@srmainwaring 
Here is the output of the build using the flag. (I have only put the error output and not the entire build log here)
```bash
you need to install pexpect with 'python -m pip install pexpect'
gmake[2]: *** [CMakeFiles/ardupilot_build.dir/build.make:70: CMakeFiles/ardupilot_build] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:168: CMakeFiles/ardupilot_build.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```
So I installed `pexpect` and tried again. This time, it was missing the `future` module. So I repeat the procedure after installing it and now I can see big red blocks across the screen. 

[Here](https://pastebin.com/fiDrMrTA) is a pastebin link of the log
Error starts from L82
And build fail logs from L418

srmainwaring, 2024-04-19:
@aarsht7 did you install `microxrceddsgen` from the ArduPilot fork (https://github.com/ArduPilot/Micro-XRCE-DDS-Gen)? The error you are seeing is because the eProsima version does not support the additional flag reported in the errors.

aarsht7, 2024-04-19:
> did you install `microxrceddsgen` from the ArduPilot fork

No, I did not. I used the docs from eProsima when I faced the error the first time.
I tried with the fork, and `ardupilot_sitl` builds without error.

Error with pkg `microxrcedds_agent` persist

closed 2024-04-19 by aarsht7

Ryanf55, 2024-04-19:
Can you please contribute a docs PR? 
I'm lost on how you skipped that step. We've had a bunch of people skip it, and I am not sure how to improve it.

Ryanf55, 2024-04-19:
Also, please review this change too:
https://github.com/ArduPilot/ardupilot_wiki/pull/5887

aarsht7, 2024-04-19:
> Can you please contribute a docs PR?

ArduPilot/ardupilot_wiki#5888