ArduPilot/ardupilot_gz #57: ros2 topic is not comming properly

PrathameshCOLLABINT, 2024-07-28:
according to documentation "https://github.com/ArduPilot/ardupilot_gz/tree/main" it should apear like /ap/battery/battery0
but after installation 3rd time still not comming that
but 2 days back my frinds lapop /ap/battery/battery etc are comming 

document i follow for installatioon is
https://github.com/ArduPilot/ardupilot_gz/tree/main
https://gazebosim.org/docs/harmonic/install/https://cdn.discordapp.com/attachments/1263794967034462260/1267039029891960893/image.png?ex=66a7558d&is=66a6040d&hm=06ea6d5875c64096939dd1e85c57077d786f490184b05db5bc15db656cdf1c90&
[image]
[image]

i cant able to figure t out  what is going wrong 

when i run iris ros2 topic echo /navsat comming properly
but when i run wildthumper_playpen.launch.py
in navsat no data showed

PrathameshCOLLABINT, 2024-07-29:
[image]
not able to find /ardupilot_dds is the issue???

Ryanf55, 2024-07-29:
GitHub is for development issues. Please ask on discord or here: https://discuss.ardupilot.org/

closed 2024-07-29 by Ryanf55