ArduPilot/ardupilot_gz #63: Add 3D Lidar to iris for usage with Bonxai mapping example

Ryanf55, 2024-11-20:
# Background

As explained in this Bonxai demo, the ardupilot_gz `iris_lidar.launch.py` assumes a 2D lidar and it must be modified. We'd like to add 3D lidar support. https://github.com/tejalbarnwal/3D-planning-with-bonxai

# Approaches considered

* rename the iris launch file to `iris_lidar.launch.py` to `iris_2d_lidar.launch.py` and duplicate the code for 3D lidar
* Add a launch argument for 2d/3d in the launch file, rename `iris_with_lidar` to `iris_with_2d_lidar` and duplicate to `iris_with_3d_lidar`
* Alternatives?

tejalbarnwal, 2024-11-20:
Hi, I prefer the third approach, where we use a base SDF file (e.g., iris_ardupilot) and build on it to add specific sensors like 2D/3D LiDAR, depth cameras, or RGB camera to create different models like `iris_with_2D_lidar`, `iris_with_depth_camera`, etc

My only concern is ensuring consistency between the model passed via the launch argument and the model included in the world files (e.g., `iris_maze.sdf`, `iris_runway.sdf`). The developer must manually ensure they match. or example, `iris_maze.sdf`
 references the iris_with_lidar model and defines its pose in the environment

A potential improvement would be to enable the independent spawning of the iris model in the world, separate from other Gazebo components, to simplify integration. However, I am unsure how this could be implemented.

Ryanf55, 2024-11-21:
> Hi, I prefer the third approach, where we use a base SDF file (e.g., iris_ardupilot) and build on it to add specific sensors like 2D/3D LiDAR, depth cameras, or RGB camera to create different models [...]

Why don't we start with just the simple option 3 with manually ensured consistency. Did you want to take this on, or have me do it?

tejalbarnwal, 2024-11-21:
I will implement it tonight after my job and submit a PR.

tejalbarnwal, 2024-11-27:
Hi @Ryanf55 ,
I have created a pull request and am still making a few minor changes, as described in the PR description.
One difference from what was discussed in point 3 is that I did not create an `iris_ardupilot` model. The basic models already exist in the `ardupilot_gazebo` repository. This repo includes models like `iris_with_ardupilot` and `iris_with_standoffs`, which do not have any perception sensors. These models are already used internally by the `iris_with_lidar` model.

Let me know your reviews for the PR.