ArduPilot/ardupilot_gz #69: Multi-vehicle simulation using ardupilot_gz

Embeida, 2025-02-26:
Hello everyone,

I want to do a simulation of a swarm of drones using ardupilot, gazebo and ros2 humble. For this purpose I have seen the package ardupilot_gz that works fine for a single drone.

I wanted to change the simulation to achieve the same with multi-drones but it appears that all the swarm ends in the same topics /ap/*

Here are the modification I did on the launch files:

(iris_runway.launch.py)->(iris_runway_swarm.launch.py)
# Copyright 2023 ArduPilot.org.
#
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program. If not, see <https://www.gnu.org/licenses/>.

# Adapted from https://github.com/gazebosim/ros_gz_project_template
#
# Copyright 2019 Open Source Robotics Foundation, Inc.
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#     http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.

"""Launch an iris quadcopter in Gazebo and Rviz."""
from pathlib import Path

from ament_index_python.packages import get_package_share_directory

from launch import LaunchDescription
from launch.actions import DeclareLaunchArgument
from launch.actions import IncludeLaunchDescription
from launch.conditions import IfCondition
from launch.launch_description_sources import PythonLaunchDescriptionSource
from launch.substitutions import LaunchConfiguration
from launch.substitutions import PathJoinSubstitution

from launch_ros.actions import Node
from launch_ros.substitutions import FindPackageShare

def generate_launch_description():
    """Generate a launch description for a iris quadcopter."""
    pkg_project_bringup = get_package_share_directory("ardupilot_gz_bringup")
    pkg_project_gazebo = get_package_share_directory("ardupilot_gz_gazebo")
    pkg_ros_gz_sim = get_package_share_directory("ros_gz_sim")

    # Iris.
    iris1 = IncludeLaunchDescription(
        PythonLaunchDescriptionSource(
            [
                PathJoinSubstitution(
                    [
                        FindPackageShare("ardupilot_gz_bringup"),
                        "launch",
[... 452 lines cut ...]
    )

    return LaunchDescription(
        [
            DeclareLaunchArgument(
                "use_gz_tf", default_value="true", description="Use Gazebo TF."
            ),
            sitl_dds,
            robot_state_publisher,
            bridge,
            RegisterEventHandler(
                OnProcessStart(
                    target_action=bridge,
                    on_start=[
                        topic_tools_tf
                    ]
                )
            ),
        ]
    )

When I launch the main file all appears to be ok appart the fact that both drones are pluged on the same topics and service /ap/*. I'm quite novice so I definitely need help to find the solution. If someone has it, it would be wonderful.

Sorry if the problem is not clear and Thank you for any help.
Embeida

TannerGilbert, 2025-02-26:
Hello @Embeida a multiagent extension to ardupilot_gz is currently in review (https://github.com/ArduPilot/ardupilot_gz/pull/67). It will allow you to create multiple vehicles using a list as follows:

```python
robots = [
        {
            "name": "drone1",
            "model": Vehicle.IRIS,
            "position": ["0.0", "0.0", "0.195", "0", "0", "1.5708"],
        },  # [x, y, z, roll, pitch, yaw]
        {
            "name": "drone2",
            "model": Vehicle.IRIS_LIDAR,
            "position": ["1.0", "0.0", "0.195", "0", "0", "1.5708"],
        },
        {
            "name": "rover1",
            "model": Vehicle.WILD_THUMPER,
            "position": ["-1.0", "0.0", "0.195", "0", "0", "1.5708"],
        },
    ]
```

Hopefully it will be merged soon but you can already use it by adding the fork as a remote and fetching the branch. Hopefully this is helpful.

Best regards,
Gilbert

PS: If possible please update the formating of the initial message to make it more readable.

Embeida, 2025-02-27:
Thank you for your extension. It is exactly what I wanted to do.

I have just one question, when you launch all the agents, the topics permitting the control by ros2 /ap/* still not duplicated, so how do you manage to control them separately?

TannerGilbert, 2025-02-28:
You can control the drones via Mavlink. This is possible graphically using for example [QGroundControl](https://qgroundcontrol.com/) or through Python for example with [pymavlink](https://github.com/ArduPilot/pymavlink)