ArduPilot/ardupilot_gz #74: `ap/geopose/filtered` doesn't reflect global position set in Gazebo world

TannerGilbert, 2025-05-23:
Gazebo allows us to set the global position of a world via:
```
<spherical_coordinates>
  <surface_model>EARTH_WGS84</surface_model>
  <world_frame_orientation>ENU</world_frame_orientation>
  <latitude_deg>37.412173071650805</latitude_deg>
  <longitude_deg>-121.998878727967</longitude_deg>
  <elevation>38</elevation>
</spherical_coordinates>
```

unfortunately this is not reflected by `ap/geopose/filtered` or MissionPlanner/QGroundControl but works on the topics from the `navsat_sensor` sensors.

srmainwaring, 2025-05-23:
This is because the Gazebo `navsat` sensor does not pass any information back to the FC. It's not really an issue with this repo, rather the data needs to be supplied to the FC from Gazebo, either through the `ardupilot_gazebo` repo `ArduPilotPlugin` or via another mechanism. My preferred approach would be to use DroneCAN following this prototype: https://discuss.ardupilot.org/t/sending-gazebo-sensor-data-to-ardupilot-using-dronecan/125730

TannerGilbert, 2025-05-23:
Thanks for the quick reply. The integration with DroneCAN looks promising. I'll try this out with the GPS topic when I find the time.

srmainwaring, 2025-05-23:
> Thanks for the quick reply. The integration with DroneCAN looks promising. I'll try this out with the GPS topic when I find the time.

There is some interest from other devs to use this for sensor integration as well - the approach needs a configuration framework, something like the yaml config used in `gz_ros`, to make it scalable. We may also need to review some of the DroneCAN messages, or augment the information coming from Gazebo.

TannerGilbert, 2025-05-24:
As another workaround I found the `home` parameter inside the `ardupilot_sitl/launch.py`. When settings this as `"home": "46.607213,14.278461,446.0,0"` the pose of the GPS was initialized correctly but no matter what I inputed for the yaw the quad is always aligned to north:
[image]

Also disabling the compass didn't help resolve this:
[image]

srmainwaring, 2025-05-24:
@TannerGilbert - try the `--custom-location="-35.28206849440639, 149.0059138453497, 581, 0"` option when starting SITL. The last parameter is heading in degrees.